tectijuana/sensores-archived #14: Módulo Sensor de Sonido de micrófono

esdrasma, 2020-03-13:


esdrasma, 2020-03-13:
# Sensor de Sonido de Micrófono

[image]

## Información
##### Éste módulo permite la conexión de un micrófono Electret a los pines digitales de un micro controlador. El nivel de detección acústico puede ser ajustado mediante un potenció metro para sensor la intensidad del sonido.
##### El módulo cuenta con ganancia ajustable para el micrófono con el fin de obtener una mejor lectura, además cuenta con tres pines para conectarlo fácilmente a un Arduino, Microcontrolador o cualquier tarjeta de desarrollo 

## Características
* Módulo para detección de ruido. 
* Cuenta con micrófono electret y sensibilidad de detección ajustable.
* Indica salida de señal 
* Salida de la señal de canal individual 
* La señal de salida es efectiva bajo nivel 
* Cuando hay sonido, salida de nivel bajo y las luces de señalización
* Cómo puede ser utilizado para la luz de control acústico' dar sonido y la luz de alarma se trabaja con sensores fotosensibles.

## Especificaciones Tecnicas
* Tamaño: aprox. 32mm * 17mm * 15mm (largo * ancho * altura) 
* Principal Chip: LM393, micrófono de condensador electret 
* Voltaje de funcionamiento: DC 4-6V 
* Color: Azul 
* Peso neto: 3g 
* Nota : La distancia máxima de la inducción es 0,5 M

IoTeacher, 2020-04-04:
https://www.tinkercad.com/circuits

`Void `

closed 2021-06-03 by IoTeacher