tectijuana/sensores-archived #32: Módulo Zumbador pasivo pequeño

VenegasMissael, 2020-03-13:


pamdmart, 2020-03-13:
Hey!

VenegasMissael, 2020-03-18:
# #32 Módulo Zumbador pasivo pequeño

![alt text][logo]

[logo]: https://www.cdmxelectronica.com/wp-content/uploads/AR0026-Sensor-KY-006-Zumbador-Buzzer-2.jpg

Se trata de un sensor piezoeléctrico el cual puede reproducir tonos de sonido los cuales están en un rango de 1.5 a 2.5 kHz. El cual se emite un sonido dependiendo de la intensidad de la variación de las frecuencias utilizadas.

## ¿Cómo conectarlo?

![alt text][conexion]

[conexion]: http://cursoarduino.proserquisa.com/wp-content/uploads/2016/10/Arduino_KY-006_Keyes_Passive_buzzer_module_connection_diagram.png

## Especificaciones técnicas

* Voltaje de Funcionamiento 1.5 – 5V DC
* Rango de generación de tonos 1.5 – 2.5 kHz
* Dimensiones: 1.8mm x 15mm
* 3 Pines
1.-(-) GND
2.- VCC
3.-(s) señal
* Corriente de trabajo: < 25mA

## Precios

| **Establecimiento** | **Precio** |
|---------------------|------------|
| Amazon              | $76 pesos  |
| Steren              | $48 pesos  |
| Ebay                | $8 pesos   |
| CDMX Electronica    | $25 pesos  |

## Ejemplo de código
En este código de ejemplo, se utilizará un Arduino para reproducir el tema de Piratas del Caribe utilizando el zumbador pasivo.

```void setup(){
 pinMode (9, OUTPUT); //pin configurado como salida
}
// funcion = tone(pin, frequency, duration)
void loop() {
  tone(9,293.66,200);
  delay(200);
  tone(9,293.66,100);
  delay(100);
  tone(9,293.66,200);
  delay(200);
  tone(9,293.66,100);
  delay(100);
  tone(9,293.66,200);
  delay(200);
  tone(9,293.66,100);
  delay(100);
  tone(9,293.66,100);
  delay(100);
  tone(9,293.66,100);
  delay(100);
  tone(9,293.66,100);
  delay(100);
  tone(9,293.66,200);
  delay(200);
  tone(9,293.66,100);
  delay(100);
  tone(9,293.66,200);
  delay(200);
  tone(9,293.66,100);
  delay(100);
  tone(9,293.66,200);
  delay(200);
[... 294 lines cut ...]
  tone(9,587.33,100);
  delay(100);
  tone(9,440,100);
  delay(200);
  tone(9,587.33,100);
  delay(300);
  tone(9,554.36,100);
  delay(400);
  tone(9,1567.98,100);
  delay(100);
  tone(9,1567.98,100);
  delay(100);
  }
```

## Diferencias entre zumbador activo y pasivo.
El zumbador activo y pasivo, a simple vista parecen idénticos pero mas de cerca la altura de los dos es ligeramente diferente, el zumbador activo es de una altura de 9 mm, mientras que el pasivo es de 8 mm.

El zumbador activo genera un tono usando un oscilador interno, por lo que todo lo que se necesita es un voltaje de Corriente continua (CC). Un zumbador pasivo requiere una señal de Corriente Alterna (CA) para Hacer un sonido.

## Ejemplos en vídeo.

[image]

[image]

VenegasMissael, 2020-03-18:
Agregado descripción, precios, ejemplos.

closed 2021-06-03 by IoTeacher