tectijuana/sensores-archived #8: Módulo de sensor de temperatura y humedad

elbk201, 2020-03-13:


elbk201, 2020-03-13:
**Modulo de Sensor de temperatura y humedad**

**Modelo DHT11**
Es un sensor de temperatura y humedad relativa de media precisión, este modulo usa un termistor NTC y un sensor capacitivo de humedad para determinar las condiciones del entorno, tiene un tamaño ultra compacto, es de bajo consumo de energía y tiene gran utilidad cuando se requiere detectar dos magnitudes al mismo tiempo, se utiliza en un gran número de aplicaciones básicas.

Te permitirá disponer de todas las funciones que proveé un DHT11 (salida de señal digital de temperatura y analógica de humedad) pero gracias a su PCB te garantiza una mayor fiabilidad y estabilidad a largo plazo, además de que ahora es más fácil adaptarlo a tu proyecto ya que el número de conexiones está minimizado un 25% comparado a un DHT11 simple.

[image]

Este módulo tiene un alto rendimiento de 8-bits; incluye un elemento resistivo y una sensación de mojado NTC dispositivos de medición de temperatura. Tiene una excelente calidad, rapidez de respuesta, la capacidad anti-interferencia y ventajas de rendimiento. Incluye LED indicador de funcionamiento y Header macho hembra para apoyar tu conexión.

**Especificaciones**

- Material: DHT11 Plástico + PCB + Dupont
- Detectar la humedad y la temperatura ambiente
- Rango de humedad de detección: 20% ~ 95% + / – 5%
- Rango de detección de temperatura: de 0 ° C ~ 50’C + / – 2’C
- Voltaje de funcionamiento: 3.3V ~ 5V
- Tipo de salida: Salida digital
- Con indicador de encendido (rojo)
- 3 Pines
- VCC: External 3.3V ~ 5V
- GND: External GND
- DO: Pequeño puerto conmutador tarjeta digital de salida, conectado al microcontrolador IO puerto
- Dimensiones: 1,50 x 0,55 en en x 0,28 en (3,8 cm x 1,4 cm x 0,7 cm)
- Peso: 0,18 oz (5 g)

**Como transmite datos?**
Aunque lo conectemos a un pin digital, se trata de un dispositivo analógico. Dentro del propio dispositivo se hace la conversión entre analógico y digital.

Por lo tanto, partimos de una señal analógica que luego es convertida en formato digital y se enviará al microcontrolador. La trama de datos es de 40 bits correspondiente a la información de humedad y temperatura del DHT11.
[image]

El primer grupo de 8-bit es la parte entera de la humedad y el segundo grupo la parte decimal. Lo mismo ocurre con el tercer y cuarto grupo, la parte entera de la temperatura y la parte decimal. Por último los bits de paridad para confirmar que no hay datos corruptos.

Estos bits de paridad lo único que hacen es asegurarnos de que la información es correcta, sumando los 4 primero grupos de 8-bit. Esta suma debe ser igual a los bit de paridad. Si nos centramos en la imagen anterior y sumamos los bits, comprobamos que todo está correcto.

**Conectado a Arduino**
Como tenemos dos versiones del sensor de temperatura y humedad DHT11, vamos a ver cómo tendríamos que conectarlo a Arduino en los dos casos. Te recuerdo que vamos a trabajar con un pin digital.

En los dos casos , la alimentación puede ser de 3,5 V a 5 V. Si vas a utilizar un Arduino MKR1000 o un ESP8266, tendrás que tenerlo en cuenta ya que estos dos dispositivos pueden dar problemas si se alimentan con una batería o con una pila ya que no suministran más de 3,3 V.

[image]

uliomar87, 2021-04-30:
> _No description provided._

closed 2021-04-30 by uliomar87

uliomar87, 2021-04-30:
**Modulo de Sensor de temperatura y humedad**

**Modelo DHT11**
Es un sensor de temperatura y humedad relativa de media precisión, este modulo usa un termistor NTC y un sensor capacitivo de humedad para determinar las condiciones del entorno, tiene un tamaño ultra compacto, es de bajo consumo de energía y tiene gran utilidad cuando se requiere detectar dos magnitudes al mismo tiempo, se utiliza en un gran número de aplicaciones básicas.

Te permitirá disponer de todas las funciones que provee un DHT11 (salida de señal digital de temperatura y analógica de humedad) pero gracias a su PCB te garantiza una mayor fiabilidad y estabilidad a largo plazo, además de que ahora es más fácil adaptarlo a tu proyecto ya que el número de conexiones está minimizado un 25% comparado a un DHT11 simple.

[image]

Este módulo tiene un alto rendimiento de 8-bits; incluye un elemento resistivo y una sensación de mojado NTC dispositivos de medición de temperatura. Tiene una excelente calidad, rapidez de respuesta, la capacidad anti-interferencia y ventajas de rendimiento. Incluye LED indicador de funcionamiento y Header macho hembra para apoyar tu conexión.

**Especificaciones**

- Material: DHT11 Plástico + PCB + Dupont
- Detectar la humedad y la temperatura ambiente
- Rango de humedad de detección: 20% ~ 95% + / – 5%
- Rango de detección de temperatura: de 0 ° C ~ 50’C + / – 2’C
- Voltaje de funcionamiento: 3.3V ~ 5V
- Tipo de salida: Salida digital
- Con indicador de encendido (rojo)
- 3 Pines
- VCC: External 3.3V ~ 5V
- GND: External GND
- DO: Pequeño puerto conmutador tarjeta digital de salida, conectado al microcontrolador IO puerto
- Dimensiones: 1,50 x 0,55 en en x 0,28 en (3,8 cm x 1,4 cm x 0,7 cm)
- Peso: 0,18 oz (5 g)

**Como transmite datos?**
Aunque lo conectemos a un pin digital, se trata de un dispositivo analógico. Dentro del propio dispositivo se hace la conversión entre analógico y digital.

Por lo tanto, partimos de una señal analógica que luego es convertida en formato digital y se enviará al microcontrolador. La trama de datos es de 40 bits correspondiente a la información de humedad y temperatura del DHT11.
[image]

El primer grupo de 8-bit es la parte entera de la humedad y el segundo grupo la parte decimal. Lo mismo ocurre con el tercer y cuarto grupo, la parte entera de la temperatura y la parte decimal. Por último los bits de paridad para confirmar que no hay datos corruptos.

Estos bits de paridad lo único que hacen es asegurarnos de que la información es correcta, sumando los 4 primero grupos de 8-bit. Esta suma debe ser igual a los bit de paridad. Si nos centramos en la imagen anterior y sumamos los bits, comprobamos que todo está correcto.

**Conectado a Arduino**
Como tenemos dos versiones del sensor de temperatura y humedad DHT11, vamos a ver cómo tendríamos que conectarlo a Arduino en los dos casos. Te recuerdo que vamos a trabajar con un pin digital.

En los dos casos , la alimentación puede ser de 3,5 V a 5 V. Si vas a utilizar un Arduino MKR1000 o un ESP8266, tendrás que tenerlo en cuenta ya que estos dos dispositivos pueden dar problemas si se alimentan con una batería o con una pila ya que no suministran más de 3,3 V.

[image]