tectijuana/sensores-archived #39: Módulo: LCD con o sin interface

IoTeacher, 2020-03-17:
Introduction As we all know, though LCD and some other displays greatly enrich the man-machine interaction, they share a common weakness. When they are connected to a controller, multiple IOs will be occupied of the controller which has no so many outer ports. Also it restricts other functions of the controller. Therefore, LCD1602 with an I2C port is developed to solve the problem. It is also simple to handle. It has only two bidirectional data lines, a serial data line and a serial clock one.
How to wire: GND—GND; VCC—VCC; SDA—A4; SCL—A5
Application Used to display Characters, letters and Numbers.
Component:1 * I2C LCD1602

IoTeacher, 2020-03-17:
IIC I2C TWI Serial 2004 20x4 LCD Module Shield for Arduino Uno Mega2560
earlier in 2017. The I2C interface promised a much cleaner interface than the standard one.
Well worth the price difference but initially I was at a loss to find drivers and example code to drive it.
Finially I discovered this code fragment that initializes the board. It works with version 1.1.2 of
the LiquidCrystal_I2C library by Frank de Brabander which is included in version 1.8.2 of the Arduino IDE
```

#include <Wire.h>
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27,20,4);

void setup()
{
Wire.begin();
lcd.init();
lcd.backlight();

}
```

IoTeacher, 2020-03-17:
Color: 20x4 I2C LCDVerified Purchase
This worked perfectly when connecting to a Particle Photon. I2C Address is 0x27, and I used 5K pull-up resistors on the SDA and SCI lines. The display is nice and bright. It was seriously easy to use.

Below is the test program I used from Particle (search the Particle.IO library for LiquidCrystal_I2C_Spark.h":

```
/*
8-Feb-2015
Jim Brower
bulldoglowell@gmail.com
*/

#include "LiquidCrystal_I2C_Spark.h"

LiquidCrystal_I2C *lcd;

int lastSecond = 0;

void setup(void)
{
Serial.begin(9600);
lcd = new LiquidCrystal_I2C(0x27, 16, 4);
lcd->init();
lcd->backlight();
lcd->clear();
lcd->print("***Spark Time***");
}

void loop(void)
{
if (Time.second() != lastSecond)
{
Serial.print(Time.timeStr());
lcd->setCursor(0,1);
lcd->print(Time.hour() < 10? " 0" : " ");
lcd->print(Time.hour());
lcd->print(Time.minute() < 10? ":0": ":");
lcd->print(Time.minute());
lcd->print(Time.second() < 10? ":0": ":");
lcd->print(Time.second());
lastSecond = Time.second();
}
}
```

IoTeacher, 2020-03-17:
https://www.amazon.com/JANSANE-Arduino-Display-Interface-Raspberry/dp/B07D83DY17/ref=pd_sbs_147_1/139-4088737-5596051?_encoding=UTF8&pd_rd_i=B07D83DY17&pd_rd_r=4e5851a1-0890-4316-a73d-ce5943294e26&pd_rd_w=KsYB1&pd_rd_wg=8fcPh&pf_rd_p=7cd8f929-4345-4bf2-a554-7d7588b3dd5f&pf_rd_r=6JAEPD3S9E50KZM3W7CT&refRID=6JAEPD3S9E50KZM3W7CT&th=1

SolisLaredoPerezGerardoAlberto, 2020-03-27:
> [image]
> 
> Introducción Como todos sabemos, aunque las pantallas LCD y algunas otras enriquecen enormemente la interacción hombre-máquina, comparten una debilidad común. Cuando están conectados a un controlador, se ocuparán varias E / S del controlador que no tiene tantos puertos externos. También restringe otras funciones del controlador. Por lo tanto, LCD1602 con un puerto I2C está desarrollado para resolver el problema. También es simple de manejar. Tiene solo dos líneas de datos bidireccionales, una línea de datos en serie y una de reloj en serie.
> Cómo cablear: GND — GND; VCC — VCC; SDA — A4; SCL:
> aplicación A5 Se utiliza para mostrar caracteres, letras y números.
> Componente: 1 * I2C LCD1602
https://www.google.com.mx/url?sa=i&url=https%3A%2F%2Fwww.naylampmechatronics.com%2Fblog%2F35_Tutorial--LCD-con-I2C-controla-un-LCD-con-so.html&psig=AOvVaw29xx45SXNexc1hg8883ZkR&ust=1585367900259000&source=images&cd=vfe&ved=0CAIQjRxqFwoTCMiY2OzhuegCFQAAAAAdAAAAABAD

IoTeacher, 2020-03-29:
[image]

[LCDRS2.pdf](https://github.com/tectijuana/sensores/files/4400057/LCDRS2.pdf)

IoTeacher, 2020-03-29:
[LCDRS2.pdf](https://github.com/tectijuana/sensores/files/4400044/LCDRS2.pdf)

closed 2020-03-29 by IoTeacher