tectijuana/sensores-archived #65: Sensor infrarojos

DianaHFer, 2021-05-07:


DianaHFer, 2021-05-07:
### SENSOR INFRAROJO.
Dispositivo electrónico/mecánico/químico que mapea un atributo ambiental resultando una medida cuantizada, normalmente un nivel de tensión eléctrica. Particularmente, el sensor infrarrojo es un dispositivo electrónico capaz de medir la radiación electromagnética infrarroja de los cuerpos en su campo de visión.

[image]

CesarAm233, 2021-12-04:
[image]

Un detector infrarrojo es un tipo de dispositivo que tiene la capacidad de medir la radiación electromagnética infrarroja que emiten los cuerpos que se encuentran dentro de su campo de visión. Se trata de un tipo de radiación que emiten todos los cuerpos de forma independiente a que exista otro tipo de luz ambiental. De este modo, permite observar espacios y objetos sin necesidad de que exista luz visible o de otro tipo en el entorno.

[image]

```python
#include <IRremote.h>
#define pinSensorIR 11

IRrecv sensorIR(pinSensorIR) ;
decode_results codigoIR;

void setup() {
	Serial.begin(9600);
	sensorIR.enableIRIn();
}

void loop () {
	if (sensorIR.decode(&codigoIR)) {
		Serial.println(codigoIR.value,HEX);
		sensorIR.resume () ;
	}
	delay (100);
}
```

IoTeacher, 2021-12-15:
Pueden ser tan amables de mover el codigo a CircuitPython

closed 2021-12-15 by IoTeacher