tectijuana/sensores-archived #82: HEARTBEAT KY-016

DanielAquinoIng, 2021-11-24:


DanielAquinoIng, 2021-12-02:
### ¿Qué es el Detector de Ritmo Cardíaco?

El Detector de Ritmo Cardíaco KY-039 es un sensor óptico de pulsos cardíacos por medio de la huella digital utiliza un LED infrarrojo brillante (IR) y un fototransistor para detectar el pulso del dedo, un LED rojo flashea con cada pulso.

Es un dispositivo de “Plug and Play”, puede ser utilizado para obtener fácilmente una lectura del ritmo cardíaco en tiempo real. 

### ¿Para que funciona el KY-039? 🤔
El KY-039 es una alternativa económica para medir el ritmo cardiaco y es compatible con diversas tarjetas de desarrollo como Arduino, Raspberry, ESP82, entre otras.

### ESPECIFICACIONES TÉCNICAS.
- Modelo: KY-039
- Voltaje: 3.3V a 5 V
- Tipo de Salida: Analógica
- Peso: 4 g
- Dimensiones: 25mm x 12mm x 12 mm
[image]

### ¿Cómo funciona el sensor de pulso cardíaco?
El monitor de pulso funciona de la siguiente manera: el LED es el lado luminoso del dedo y el fototransistor del otro lado del dedo, el fototransistor utilizado para obtener el flujo emitido, cuando la presión sanguínea pulsa con el dedo cuando la resistencia del fototransistor ser ligeramente cambiado.

Cuando el pulso de la presión arterial pasa por el dedo la corriente de la base del fototransistor se modifica ligeramente, debido a que la cantidad de luz que puede atravesar el dedo disminuye, lo que significa una salida por el puerto análogo diferente.

Es muy importante tener en cuenta que es prioritario que se mantenga lo más alejado de la luz parásita, es decir de la luz externa por ejemplo la iluminación del hogar, por eso se recomienda usar algún sistema que sirva de escudo al fototransistor y que evite que otra luz diferente a la del led llegue ya que la señal del latido del corazón es muy débil y la luz ambiente añadirá un ruido considerable.

DanielAquinoIng, 2021-12-03:
### Ficha técnica

[image]

Este modulo tiene tres pins: GND, Vcc+ y Signal. El pinout es el siguiente:
[image]

El esquema del sensor de latido KY-039 es el siguiente:
[image]
https://www.thegeekpub.com/wp-content/uploads/2019/07/KY-039-heartbeat-sensor-schematic.jpg

DanielAquinoIng, 2021-12-03:
## Modo de conexión 

| Pico   | KY-039 Heartbeat |
|--------|------------------|
| Pin 38 | GND              |
| Pin 36 | 3.3V a 5V        |
| Pin 27 | Señal analógica  |

# Diagrama
[image]

_**El diagrama fue realizado en Fritzing, no encontré la librería para el sensor. Encontré uno similar y le coloque etiquetas para identificar cada parte**_

## Sensor y salidas: 
[image]

DanielAquinoIng, 2021-12-03:
# Código

```python
# Aquino Villegas Daniel 18212144
# No logré que arroje valor

import machine
import time

factor = 0.75
maximo = 0.0
minimoEntreLatidos = 300
valorAnterior = 500
valorverdadero = 0

sensor = machine.ADC(27)

print('iniciando mediciones')

while True:
    valorticks = time.ticks_ms()
    valorverdadero =  valorverdadero + (time.ticks_ms() - (valorticks - 1))
    
    
    tiempoLPM = valorverdadero
    entreLatidos = valorverdadero
    
    valorLeido = sensor.read_u16()
    
    valorFiltrado = factor * valorAnterior + (1 - factor) * valorLeido
    cambio = valorFiltrado - valorAnterior
    
    valorAnterior = valorFiltrado
    

     
    if((cambio >= maximo) & (valorverdadero > entreLatidos + minimoEntreLatidos)):
        maximo = cambio
        entreLatidos = valorverdadero
        latidos = latidos + 1

    
    
    maximo = maximo *0.97
    
    if (valorverdadero >= tiempoLPM + 15000):
        print('Latidos por minuto: ')
        print(latidos * 4)
        latidos = 0;
        tiempoLMP = valorverdadero
        
    time.sleep_ms(50)

Rulas-Boy, 2021-12-10:
@DanielAquinoIng 
La razon por la que no despliega nada es porque la condicion de la linea 43 nunca se va a cumplir puesto que ambos valores incrementan igual conforme el ciclo itera.

Creo que hay algo que no deberia estar dentro del ciclo infinito.

[image]

closed 2023-08-18 by IoTeacher