tectijuana/sensores-archived #89: KY-010 Light Blocking

alfredo70fil, 2021-11-25:


SalasJ13, 2021-12-02:
Marco teorico.

SalasJ13, 2021-12-02:
### Marco Teorico

SalasTec13, 2021-12-02:
#Marco teórico.
#Diagramas y tablas técnicas.
#Código
#Ejecución o comprobación.

SalasJ13, 2021-12-03:
### 🅺🆈-010 𝕊𝕖𝕟𝕤𝕠𝕣 𝕕𝕖 𝕓𝕝𝕠𝕢𝕦𝕖𝕠 𝕕𝕖 𝕝𝕦𝕫 𝕣𝕠𝕥𝕒

SalasTec13, 2021-12-03:
### 🅺🆈-010 🆂🅴🅽🆂🅾🆁 🅳🅴 🅱🅻🅾🆀🆄🅴🅾 🅳🅴 🅻🆄🆉 🆁🅾🆃🅰

SalasTec13, 2021-12-03:
[image]

SalasTec13, 2021-12-03:
### [KY-010] Sensor de bloqueo de luz rota

SalasTec13, 2021-12-03:
###**Marco Teórico.**

SalasTec13, 2021-12-03:
[image]

SalasTec13, 2021-12-03:
### **Marco Teórico**
EL ky-010 es un fotosensor que combina dos partes las cuales son transmisión y de recepción en un módulo, ya que de un lado tiene el emisor y por el otro extremo tiene un rector que recibe la señal y manda una señal cuando se interrumpe la luz infrarroja y manda una señal al microcontrolador.

SalasTec13, 2021-12-03:
### `2. Diagrama`

SalasTec13, 2021-12-03:
### `3.  Código`

# Los módulos necesarios serán importados y configurados
importar RPi.GPIO como GPIO
tiempo de importación
  
GPIO.setmode (GPIO.BCM)
  
# El pin de entrada que está conectado con el sensor
GPIO_PIN = 21
Configuración de GPIO (GPIO_PIN, GPIO.IN, pull_up_down = GPIO.PUD_DOWN)
  
imprimir "Prueba del sensor KY-010 [presione ctrl + c para finalizar la prueba]"
  
def función de salida (nulo):
         imprimir ("El sensor está bloqueado")
  
# detección de señal (flanco ascendente).
GPIO.add_event_detect (GPIO_PIN, GPIO.RISING, callback = outputFunction, tiempo de rebote = 100)
  
# Bucle del programa principal
tratar:
         mientras que es cierto:
                 hora de dormir (1)
  
# Trabajo de recolección de residuos una vez finalizado el programa
excepto KeyboardInterrupt:
         GPIO.cleanup ()

SalasJ13, 2021-12-04:
[image]

SalasTec13, 2021-12-04:
[image]

SalasTec13, 2021-12-04:
### `1. Marco Teórico`
El módulo KY-010  consta de un emisor / detector óptico en la parte frontal y dos resistencias (1 kΩ y 33 Ω) en la parte posterior. 
Es un fotosensor que combina dos partes las cuales son transmisión y de recepción en un módulo, ya que de un lado tiene el emisor y por el otro extremo tiene un rector que recibe la señal y cuando se interrumpe la luz infrarroja esta manda una señal al microcontrolador. 

> ¿Para qué es?
Se utilizan para detectar la velocidad de un giro de un motor con un una lamida también puede detectar cualquier objeto que interrumpa la luz infrarroja se puede utilizar para el cierre de una puerta con una sencilla programación y para algunos desplazamientos de máquinas.

> ¿Cómo Funciona?
El sensor ky-010 es un dispositivo electrónico que responde cuando se interrumpe la luz infrarroja el cual se manda al microcontrolador la señal se emplean en algunas aplicaciones como pueden ser en la automatización de oficinas, árcades, detección de objetos en general.

[image]

SalasTec13, 2021-12-04:
### `2. Diagrama`

[image]

> Especificaciones Técnicas.
- Voltaje de funcionamiento 3.3 ~ 5V.
- Dimensiones 18.5mm x 15mm [0.728in x 0.591in]

> Diagrama de conexión KY-010.
[image]

> Conexiones.
| KY-010    	| Rasperry        	|
|-----------	|-----------------	|
| (-)       	| (Izquierda) GND 	|
| Pin medio 	| + 5V            	|
| (S)       	| (Derecha) Pin 3 	|

SalasTec13, 2021-12-04:
``` python
# Los módulos necesarios serán importados y configurados

import RPi.GPIO as GPIO
import time
  
GPIO.setmode(GPIO.BCM)
  
# El pin de entrada que está conectado con el sensor

GPIO_PIN = 21
GPIO.setup(GPIO_PIN, GPIO.IN, pull_up_down = GPIO.PUD_DOWN)
  
print "KY-010 Sensor Test [press ctrl+c to end the test]"
  
def outputFunction(null):
        print("Sensor is blocked")
  
# detección de señal (flanco ascendente).

GPIO.add_event_detect(GPIO_PIN, GPIO.RISING, callback=outputFunction, bouncetime=100) 
  
# Bucle del programa principal

try:
        while True:
                time.sleep(1)
  
# Trabajo de recolección de residuos una vez finalizado el programa

except KeyboardInterrupt:
        GPIO.cleanup()
```

IoTeacher, 2022-03-10:
SE sube el cooltext por errores..
[borrame.pdf](https://github.com/tectijuana/sensores/files/8219503/borrame.pdf)

.

closed 2023-08-18 by IoTeacher