tectijuana/sensores-archived #110: (LIBRE) KY-011 Two-color

josue0620, 2021-12-07:


LuisDeJesusR, 2021-12-10:
# DIAGRAMAS
### Conexión
| KY-011 | Tablero | Pi Pico |
|--------|---------|---------|
| -      |         | Pin 38  |
| medio  | 330 Ω   | Pin 27  |
| S      | 330 Ω   | Pin 26  |

*Nota: se recomienda utilizar resistencias de 330 Ω*

[image]

LuisDeJesusR, 2021-12-10:
# TABLAS TÉCNICAS
### Especificación y características
| Propiedad                 | Característica            |
|---------------------------|---------------------------|
| Tipo Led                  | bicolor cátodo común 5 mm |
| Voltaje de funcionamiento | 2 V a 2.5 V               |
| Corriente                 | 10mA                      |
| Angulo de visión          | 150°                      |
| Longitud de onda          | 571nm a 644 nm            |
| Intensidad luminosa       | 20-40 y 40-80 MCD         |
| Color                     | Rojo y Verde              |
| Dimensiones               | 40 mm ,15 mm y 7 mm       |
| Peso                      | 1 gr                      |

### Pines
[image]
| Pin del componente | Propiedad |
|--------------------|-----------|
| (-)                | GND       |
| medio              | rojo      |
| S                  | verde     |

LuisDeJesusR, 2021-12-10:
Link del repositorio:
https://github.com/tectijuana/sensores/tree/master/PYTHON/KY-011%20Two-color

closed 2021-12-10 by DanielAquinoIng