tectijuana/sensores-archived #111: KY-034 7 Color Flash

DanielAquinoIng, 2021-12-07:


FernandoChD, 2021-12-10:
## MARCO TEÓRICO

### Descripción
El Led Flash de 7 colores es útil para sistemas en donde se requiera algún tipo de señalización visual o decorativo, por ejemplo en maquinaria, juguetes, sistemas de control, por mencionar algunos.

FernandoChD, 2021-12-10:
## DIAGRAMAS

## TABLAS TÉCNICAS

### Especificaciones

* Voltaje de funcionamiento: 3 a 5 V
* Modelo del producto: YB-3120B4PnYG-PM
* Forma: Redondo Tipo de LED DIP 5mm
* Grupo de colores: verde, amarillo y rosa (alto brillo).
* Tipo de lente: niebla blanca
* Dimensiones: 19mm x 17mm x 20 mm
* Peso: 2 g

[image]

FernandoChD, 2021-12-10:
## CÓDIGO

```python
#Modulo KY-034 7 Color Flash en Raspberry Pi Pico
#Chavez Duarte Fernando

from machine import Pin
from utime import sleep

led = Pin(3,Pin.OUT)

while True:
        led.on()
        sleep(1)
        led.off() 
        sleep(1)

closed 2021-12-10 by DanielAquinoIng

Rulas-Boy, 2021-12-16:
Estoy teniendo inconvenientes para correr el código de este módulo.
Y no sabría distingir si es por error en la codificación, o porque está quemado mi sensor.