tectijuana/sensores-archived #115: Sensor Infrarojo

IoTeacher, 2021-12-15:
Circuitpython y hacer el archivo a ejecutar

Rulas-Boy, 2021-12-15:
Me encuentro un poco ocupado profesor, @IoTeacher.
Iré directo a probar lo que comentó en Classroom en cuanto antes, aproximadamente en una hora.

CesarAm233, 2021-12-16:
[image]

CesarAm233, 2021-12-16:
Un detector infrarrojo es un tipo de dispositivo que tiene la capacidad de medir la radiación electromagnética infrarroja que emiten los cuerpos que se encuentran dentro de su campo de visión. Se trata de un tipo de radiación que emiten todos los cuerpos de forma independiente a que exista otro tipo de luz ambiental. De este modo, permite observar espacios y objetos sin necesidad de que exista luz visible o de otro tipo en el entorno.

CesarAm233, 2021-12-16:
[image]

CesarAm233, 2021-12-16:
|         Proyecto       |      Código    |    
|-----------------------|---------------|
|  Control remoto          | C++ | 

Por el momento estaré buscando su otra version de MicroPhyton / CircuitPython.
Seria de gran ayuda si uno de los tester me den una mano con ello.

CesarAm233, 2021-12-16:
```python
#include <IRremote.h>
#define pinSensorIR 11

IRrecv sensorIR(pinSensorIR) ;
decode_results codigoIR;

void setup() {
	Serial.begin(9600);
	sensorIR.enableIRIn();
}

void loop () {
	if (sensorIR.decode(&codigoIR)) {
		Serial.println(codigoIR.value,HEX);
		sensorIR.resume () ;
	}
	delay (100);
}
```